ntrteam/ntrboot_flasher #1: 2011 Acekard 2i not supported.

Cuphat, 2017-08-12:
When trying to dump the flash of my old AK2i bought in 2011, it shows:

ChipID: EA17C4FF
HW Rev: EA17C4FF
Supported cartridge not detected!

This cart otherwise seems to work fine, AKAIO 1.9.0, AK2i updates show it as HW-81.

A new AK2i that I purchased just this year seems to be detected just fine though.

Normmatt, 2017-08-12:
That sounds like a contact issue... try re-seating the cart a few times.

d3m3vilurr, 2017-08-12:
https://wiki.gbatemp.net/wiki/Solving_Acekard_2i_Contact_Issues please check either :)

Cuphat, 2017-08-12:
Thanks for confirming that it could be a contact issue. The cart does have some contact issues, but I have re-seated it a number of times, confirmed that the 3DS menu and GodMode9 can detect it before trying, and it still works great as a normal flashcart, so I figured there might be more to it.

Committee, 2017-08-12:
Same issue here.
[image]

pixelmasher, 2017-08-12:
I have a similar issue. I'm not sure what year my card is, but I got it when the Acekard team was still around. When I eject and reinstert the card, it sometimes shows different Chip ID and HW Rev. It seems to be random. I've seen several different ones. Tried about 20 times on 2DS.

Tried on N3DS. It had different results. It detected the card as EA17C4FF about half a dozen times, once it detected it as different Chip ID, but same HW Rev. It gave the not supported card message, but worked after about half a dozen tries. It wouldn't inject, but it worked on second try.

I have two cards both have no issues being recognized, dumping in Decrypt9, or playing NDS files on several 2DS/N3DS. It's still buggy.

The injected card doesn't work. Gave up after a while.

It would be great if there is a way to check the injected card to make sure it was properly injected.

Cuphat, 2017-08-12:
I just used d3m3vilurr's Decrypt9 branch and ak2ifirm generator tool to flash my old AK2i and it works perfectly. I injected my new AK2i with ntrboot_flasher and it would not work as an ntrboot card (the 2DS just booted straight to stock firmware).

¯\_(ツ)_/¯

All that said, thanks everybody. This is an exciting release.

urherenow, 2017-08-12:
@Cuphat I believe this issue should be fixed with the flashcart_core commit. Either way, mind linking to this ak2ifirm generator?

dragonkni, 2017-08-12:
I got acekard HW81 and akaio 1.9.0. is working fine.
when i try to flash ntrboothax i got messege 
ChipID: EA17C4FF (sometimes this id is different)
HW Rev: EA17C4FF  (sometimes this HW is different)
What i need to do? I tried:
Cleaning , removing microsd, i dont know how to build another version of ntrboot_flasher....
Help?

KaschWarckaterus, 2017-08-12:
I have an AK2i HW81 on kernel 1.41 (with already inserted paper to fix the contact issues) and I could dump it seemingly without problems and also flashing it succedded as far as it seems (Even though dumping took a really short time).
But in the home menu the acekard icon is displayed and also the functions work like before, the ntrboot-hax doesn't work however.
I placed the magnet on my New 3DS no XL below and left of the b-button were it works if it is turned on, so it should work from coldboot too I guess...

Arch9SK7, 2017-08-12:
I am getting the exact same issue and i know it isnt contact issue. but i cant dump the flash using the forked D9WIP due to me being in b9S already.

Anyone figure a real working fix for the issue? My Acekard 2i says DSI(XL) & 3DS Compatible. Was that even one of the HW revisions that were compatible?

pixelmasher, 2017-08-12:
Still not working with 0.1.1. The Acekard gets detected as Chip ID: 00000FC2 Hw Rev.: 81818181. Seems to flash fine. When inserted and used, it doesn't work. If the Start + Select + X is held, the blue light comes on, but nothing. If you let the buttons go, it goes off.

Arch9SK7, 2017-08-12:
Did you try using a magnet.......

Cuphat, 2017-08-12:
@pixelmasher You still need a boot.firm on your 3DS's SD card. It turning off when releasing the buttons sounds exactly what B9S does when it doesn't find a payload.

Committee, 2017-08-12:
Still same error for me after updating to 1.1

Arch9SK7, 2017-08-12:
@kitling Do you have a possible solution to the initial issue quite a few of us have? we have established that it isn't to do with contact.

dragonkni, 2017-08-12:
1.1 version is not detecting my hw81

kitling, 2017-08-12:
@Arch9SK7 Not at the moment.

For the people mentioning v0.1.1, this is not the issue it was meant to fix, it only fixed injection if your cart was already detected, which is not the issue happening here.

All I can say is that we must have not had a cart that displayed this issue when we were testing.

pixelmasher, 2017-08-12:
@Cuphat I was using the wrong microSD card. I forgot to switch it back to try it. I put the original in it when I put it away for the night. Still doesn't work. Maybe I'll try that file you posted on temp.

@kitling I think the detection is wonky or doesn't like certain cards. I have 2 Acekard 2i and one seems to get detected fine most of the time (horse icon), but the other one (downgraded to 1.41 since @d3m3vilurr mentioned that firm) that I'm actually trying to use gets detected as usually Chip ID: 000000FC2 Hw Rev: EA17C4FF, Chip ID: EA17C4FF Hw Rev: 81818181, or some combination thereof.

kitling, 2017-08-12:
```
<profi200> Btw: The delays in cardslot reset function are way too low. That is what causes most init fails with legit games. See GodMode9.
```

I shall look into increasing the delays and provide a test build for that in a bit, hopefully.

kitling, 2017-08-12:
People that are having issues with detection: could you try this build real quick? https://up1.secretalgorithm.com/#81b9kTBd7s3b1EDJ2S2TLw

profi200 tried it, but I'm going to have everyone here test it first before I push.

ALEXM99999, 2017-08-12:
@kitling it worked! Thank you!

dragonkni, 2017-08-12:
Kitling it worked on my acekard 2i ty!! No more problem with detection!

Arch9SK7, 2017-08-12:
@kitling can confirmed it worked 100%

kitling, 2017-08-12:
Commit c3979128da59ab3da1613e70e9abbceea1b39096 should fix this. Making a release v0.1.2 momentarilly.

closed 2017-08-12 by kitling

pixelmasher, 2017-08-12:
The card gets detected correctly on first try now.