ntrteam/ntrboot_flasher #3: R4i-SDHC RTS not working

kasianvh, 2017-08-12:
will the R4i-SDHC ever work with ntrboot?

CookieXD, 2017-08-12:
and what's with the Ace3ds Plus?

N00byKing, 2017-08-12:
I don't think Ace3DS+ will ever work because it is a CTR Card (it identifies as a 3DS, not NDS or DSi Card)
However I am no expert, you should probably ask around on the Nintendo Discord Channel or something (https://discord.gg/MWxPgEp)

chronoss09, 2017-08-12:
And nothing for R4 Dual Core RTS Lite ?

kitling, 2017-08-12:
So... is everyone here just asking when their cart will be supported? I'm afraid that I can't give an answer on that, and quite frankly, this is not what Github issues are for.

The only cards supported right now are in the readme. There's some leeway on the R4i (Gold?) 3DS (RTS?) because, frankly, I'm not quite sure which cards fall under it's category.

If you want to help with this, please find an updater that works for your flashcart (*.nds format) and open an issue on [flashcart_core](https://github.com/kitling/flashcart_core) with a link to an updater. If you can do some reverse engineering/have IDA, it would be much appreciated if you could find the cartridge commands (Read, Write, a means of identifying the cart, flashbusy, anything needed for setup) and post them in that issue.

closed 2017-08-12 by kitling

TheSitX, 2017-08-16:
Sky3ds plus , will be added?

hedgeberg, 2017-08-16:
The sky3ds cannot behave as an ntr cart, so no. Please stop asking on this repo and read the readme.