ntrteam/ntrboot_flasher #4: I think the r4idsn has the same chip like the R4i Gold 3DS RTS

pandavova, 2017-08-12:
I tryed to dump the Flash of my r4idsn and got this:
`<blockquote class="imgur-embed-pub" lang="en" data-id="a/EJonO"><a href="//imgur.com/EJonO">pandavova</a></blockquote><script async src="//s.imgur.com/min/embed.js" charset="utf-8"></script>`

kitling, 2017-08-12:
I'd say give it a try! I'm not quite sure what all carts fall under the 'R4i 3DS Gold' umbrella yet. If it doesn't work, then you *should* be able to restore the backup.

Just in case, you should find a firmware updater that works for your cart, and post it here if it doesn't work and you can't restore.

pandavova, 2017-08-12:
Done it (like some hours ago, backing up the r4idsn worked and flashing back too), installed sighax on my bricked 3DS XL, did a CTRtransfer, updated my 3DS with the recovery mode and have now a working 3DS XL again!
(But the plastic thing at the touch screen ribbon connector broke,now the colors on the touchscreen are not normal :( )

kitling, 2017-08-12:
GG! Glad that it worked for you.

closed 2017-08-12 by kitling