ntrteam/ntrboot_flasher #7: Acekard 2i not supported?

IvanMolina99, 2017-08-12:
I try several times to flash the Acekard 2i HW-44 and it inyect successfull the exploit but when i try it on a 3ds  bricked it doesn´t work. And if I open the flashcart it opens normally....

kitling, 2017-08-12:
On the bricked 3DS: do you have a boot.firm on the SD card? Are you holding Start+Select+X with a magnet in the right place before powering on? (The correct placement is harder to find with a bricked console, as you can't experiment as easily, but someone should be able to help you with that.)

Hikari-chin, 2017-08-12:
Can't reproduce. Have an Acekard 2i HW44, works fine. It's simply not possible for ntrboot to "brick" a 3DS, either.

closed 2017-08-12 by Hikari-chin

kitling, 2017-08-12:
@Hikari-chin I believe that they meant that their 3DS was *already* bricked.

Hikari-chin, 2017-08-12:
I see. Well, for clarification: 

* ntrboot should work without issue on every device with two exceptions: MCU bricked devices (only one exists as far as I'm aware) and devices that have significant hardware damage.