ntrteam/ntrboot_flasher #8: List of Fashcards covered by "R4i Gold 3DS"

kitling, 2017-08-12:
Please list your flashcart here if it was detected as an R4i Gold 3DS, and ntrboot sucessfully worked.

Currently, we have seen reports of success with:
 - r4idsn
 - R4i 3DS RTS

The hope is to have a more-or-less comprehensive list (preferably with images) so that we can point here and say "yeah, that should be supported."

adrot, 2017-08-12:
When referring to R4i 3DS RTS, it's still about the gold variant, and not the one from www.r4i-sdhc.com?

kitling, 2017-08-12:
@adrot: Not sure, @hedgeberg will upload a picture when they get a chance.

Whovian9369, 2017-08-12:
For completion sake:
I have a http://nds-card.com/ProShow.asp?ProID=149 (R4i GOLD 3DS RTS) that's detected as "R4i Gold 3DS" (`ChipID: 00000FC2` / `HW Rev: A7A7A7A7` / `Detected: R4i Gold 3DS`)

Just wanted to put the info here, since I know that it's the most suggested NDS Flashcart that I remember seeing in the #3DSHacks IRC channel.

jerbear64, 2017-08-12:
@Whovian9369 Well, that's the one being referred to, so yeah.

Whovian9369, 2017-08-12:
@jerbear64 Just wanted to add it for completeness sake, since that seems to somewhat be the point of this issue.

kitling, 2017-08-12:
Yes, that is, indeed, the point of this issue.

kaizi99, 2017-08-12:
I have just tested my R4i RTS 3DS from www.r4i-sdhc.com and it is not detected

ChipID 000007C2
HW Rev FFFFFFFF

Tested with ntrboot flasher 0.1.2

hedgeberg, 2017-08-13:
For clarity, the card linked by @Whovian9369 here:  http://nds-card.com/ProShow.asp?ProID=149  is the card officially supported. Chip ID should be be 00000FC2, HW Rev should be A7A7A7A7. Other R4 cards may work surprisingly even if they dont have a matching case, but without that specific chip ID/hw rev pair, they will not work. If you have an r4 card, its worth testing for that revision/id via the dump flash option. If you do not, it will not work, and is likely a counterfeit of some kind. This will not be fixed as we cannot support every single counterfeit.

hedgeberg, 2017-08-13:
@kitling this issue should be suitable to close, and we should add this information to the readme so folks arent confused.

adrot, 2017-08-13:
In order to assure the conflict from my side, this is the card that I have, that is why I was conflicted, and also has the legit seal. http://nds-card.com/ProShow.asp?ProID=146

Cuphat, 2017-08-13:
@adrot Yeah, that cart is deceptively similar looking to the point where I mistakenly bought one like a year ago thinking it was an r4ids.cn cart. I was mad when I saw how inferior it was. Hopefully someone will take pity on it and make it useful one day.

Banamy, 2017-08-13:
I have non RTS R4i Gold 3ds. Not supported. 

ChipID: 00000FC2
HW Rev: 00000000

hedgeberg, 2017-08-13:
That HW Rev makes it sound like this is a counterfeit cartridge. Sorry, I doubt we'll be able to make that work

Banamy, 2017-08-13:
figured. I just ordered a new one

kitling, 2017-08-13:
profi200 also has a cart with HW Rev 00000000, so I think we just need to take a look and find a different way of identifying it.

Banamy, 2017-08-13:
I do know that with my card, it couldnt be updated to a certain point because the revision wasnt the right one. In order for it to work on the 3ds, you had to open it up and cut one of the circuits. My revision lacked the circuit.

Banamy, 2017-08-13:
[image]

My revision is 6 and it was never updated.

Banamy, 2017-08-13:
[image]
My r4 cant go past 3ds 4.5, hope this helps

hedgeberg, 2017-08-13:
What is the revision listed by ntrboot flasher when dumping the flash?

Banamy, 2017-08-13:
00000000

kitling, 2017-08-13:
@Banamy Could you open an issue on [flashcart_core](https://github.com/kitling/flashcart_core/issues) with a link to a *.nds updater and as much detail about your cart as you can find? I'm trying to make that the standard place for flashcarts that don't currently work. (also, it looks like hwid 00 r4i gold 3ds is common, so it's a good one to try and support)

uberyoji, 2017-08-13:
Tried a R4 Gold Pro and R4i RTS 3DS. Both show:
ChipID: 0000072C
HW Rev: FFFFFFFF
Not supported. Must be clones I suppose.

Valeri0p, 2017-08-13:
Maybe is better to make an issue template to redirect "support for r4X" to the other repo...

ShadixAced, 2017-08-13:
Tested with R4i SDHC 3DS (ver 4.5) (HW code and ID 7F7F7F7F7F)
Said that the card wasn't supported.

Picture just below

[image]

kitling, 2017-08-14:
Alright, this issue is going nowhere quickly. People are reporting things that don't work instead of things that work.

@Valeri0p I can do that in a few days, after I get home.

closed 2017-08-14 by kitling

Ryper-maximoft, 2017-10-05:
i found this r4 card on aliexpress it says r4i gold pro so idont if its compatible,im from mexico and i whant to check a proper flashcard for this method,i apologize for the long adress but i dont find a shorther one. 

https://es.aliexpress.com/store/product/Nintendo-R4-Gold-Card-for-3DS-NDSi-NDSL-NDS-with-Card-Reader-and-Retail-Package/2336305_32831490529.html?spm=a219c.search0104.3.1.QCtQ3p&ws_ab_test=searchweb0_0,searchweb201602_5_10152_10065_10151_10068_10344_10342_10343_10340_10341_10307_10060_10155_10154_10056_10055_10054_5370015_10059_10534_10533_10532_100031_10099_10338_10339_5580015_10103_10102_10052_10053_10142_10107_10050_10051_10173_10084_10083_10080_10082_10081_10110_5590015_10111_10112_10113_10114_143_10312_10314_10078_10079_10073-10051,searchweb201603_16,ppcSwitch_2&btsid=2b9abe32-0c51-4b69-a61f-50359fb0d690&algo_expid=f2dd7db5-6877-4e12-8947-b814f4f05886-0&algo_pvid=f2dd7db5-6877-4e12-8947-b814f4f05886

rjose2002, 2017-10-05:
still not compatible yet, I went to a store in Spain and take the 3ds with the ntr-flasher and tell me that it is not supported.
http://www.nds-card.com/ProShow.asp?ProID=149 this works fine

Ryper-maximoft, 2017-10-05:
thank you i give it a shot, thanks

kitling, 2017-10-05:
This is a closed issue. It was closed because people were asking if x card supported.

#### Then you asked if X card is supported.

Locked.