ntrteam/ntrboot_flasher #9: Acekard 2i HW-44 Issue

IvanMolina99, 2017-08-12:
I have any problems injecting the exploit on the Acekard 2.i HW-44
The ntrboot_flasher seems to install correctly the exploit on the flashcart but it´s not the reality.

I can open it from the Home Menu and play games, it doesn´t inyect correctly the exploit

You can see a video with more details in the following video on Twitter

https://twitter.com/Hirobyte/status/896467719654170626

Hikari-chin, 2017-08-12:
The Acekard 2i retains its functionality as a flashcart even after ntrboot is installed. The video you posted displays the expected functionality.

closed 2017-08-12 by Hikari-chin

kitling, 2017-08-12:
@IvanMolina99 To trigger the actual exploit, you need to hold Start+Select+X and hold a magnet in the right spot.