ntrteam/ntrboot_flasher #10: Not Working Properly on Bricked N3DS JP

Bonesaww, 2017-08-13:
Having difficulties with a blue screen brick (00000400, "NAND not found error"). Attempted a hardmod downgrade on a JP New 3DS (small one) almost half a year ago and got a brick when I wrote to the NAND. Trying to restore my unmodified NAND backup didn't fix it and I figured it was done for.

I have ntrboot setup on an AK2i HW81 now. I no longer get the bootrom error while performing the exploit, but I can't seem to load any FIRMs. It'll stay on an un-lit black screen until I either pull the battery or hold the power switch. I've triple checked that all the correct files are in the right place, tried a different microSD, tried using different FIRMS, re-flashed my AK2i on two different consoles, and even went so far as to disassemble the system again to see if anything was out of place or if there was any noticeable board damage. 

Not sure at this point if it's a random issue with ntrboot_flasher or something with the hardware itself.

hedgeberg, 2017-08-13:
Not easy to judge, tbh. Have you let go of the buttons after the power light comes on? It sounds like you might be forgetting that but I'm unsure. Is there a boot.firm to load at the root of the SD card?

hedgeberg, 2017-08-13:
Also, make sure you're using the newest build, there was a fix for hw81 iirc.

kitling, 2017-08-13:
There was a fix where the buffers used for flashing weren't properly cleared to 0, affecting ak2i in general. If you're using v0.1 and not v0.1.1 or higher, then you may have ran into that, though that wouldn't explain the black screen instead of bootrom error?

Bonesaww, 2017-08-13:
Yeah I've tried all manner of releasing them asap, holding them for a prolonged period, leaving the magnet on, taking if off, etc. I've tried several different boot.firms and none of them have worked

The exploit itself is definitely triggering something, as the console just shows the bootrom error otherwise.

 I thought maybe SafeB9SInstalled was running with the display off and tried doing the correct button presses to install it, but that doesn't appear to be the case either.

Bonesaww, 2017-08-13:
Well, <0.1.2 wouldn't even let me dump the cart in the first place, so that's not it.

kitling, 2017-08-13:
okay. What do you have on your SD card for boot.firm? What firm did you flash for ntrboot? You told it to flash retail and not dev, right?

Bonesaww, 2017-08-13:
Right now, SafeB9SInstaller.firm, but I've tried Godmode9 as well for the boot.firm. Flashed the retail version of boot9strap_ntr.firm using R at the confirmation screen. I've redownloaded pretty much every necessary file in-between attempts as well, just in case.

hedgeberg, 2017-08-14:
Ok @Bonesaww there's a few issues here that we need to resolve. I hate to say that this is beginning to sound a lot like a single-point failure meaning this likely won't get resolved. That doesn't mean we won't keep trying, but i dont want to get your hopes up. 
The main thing I can think of that might be a user error would be removing the magnet. If that doesn't work and you're still getting the same issue, we need to try compiling a debugging ntrboot firm, probably gm9 compiled to run directly from ntr. If that happens to work, the issue is in your 3ds's sdmmc reads, if it doesn't it might be that either your flashcart or 3ds has significant hardware issues which would be non-trivial to diagnose or repair

Bonesaww, 2017-08-21:
I'm gonna go ahead and close this. I'm almost certain it's a hardware issue at this point.

closed 2017-08-21 by Bonesaww