ntrteam/ntrboot_flasher #11: R4i Gold 3DS (Non-RTS)

cakehonolulu, 2017-08-13:
Vendor: www.r4i-gold.eu
CChipID: 000007C2
HW Rev: 00000000

Photographies:
Card (Front):
[image]
Card (Back):
[image]
3DS ntrboot_flasher utility:
[image]

Extra Information:
3DS Shows this cartridge as: The Chronicles of Narnia: Prince Caspian (NDS Title)

Card's 'Structure':
R4.dat (File) (Maybe is the file the flashcard searches as an entrypoint?)
R4iMenu (Folder) (Flashcard's internals: Fonts, Skins, Cheats, Languages... And a R4iLoader.dat file)
moonshl2 (Folder) (A File Explorer with the ability to launch nds homebrew)
moonmemo (Folder) (Simple readme's for instructions for getting a "Notepad" on the flashcard)
homebrew (Folder) (Some kind of Messenger for the DS called 'Beup Live')

Card 'Features':
Support the Latest N3DS perfectly.
Able to memorize and restore to the previously selected game or application after restarting machine and soft reset
Sleep mode function, for power saving and longer standby time
Multi-languages are optional, they are Simplified Chinese, complicated Chinese, English, Japanese, French, German, Italian, Spainish and Dutch
Friendly interface, easy to operate
Realtime smart help windows
Integrated the latest Moonshell 2.0 Beta version
Game compatibility: 100%
No need to flash. Built-in launch Slot-2 (gba) function
Support GOLD micro sd cards (4GB、8GB、16GB、32GB)
Support DMA read and no lag while using any microSD card
Support FAT16 and FAT32. Files can be transferred with PC, PDA, Cellphone and Camera
Support clean rom and drag-drop to play. Fast loading
Support NDS and iDS games. DS games can run on any DS versions
Files are to be saved directly to TF card, copy or restoration is not required
Automatically recognise saved new game types, no need to update the database
Support moonshell and home brew
Power saving design. Enter optimized mode automatically
Double screen UI, abbreviated picture and game title display, button and touch operations
Easy to use, setting or adjustment is not required
Support Wi-Fi, rumble pack and browser expansion
UI background can be changed
4-level brightness adjustment
Support soft reset
Support AR cheat code
Support soft-reset of moonshell 2.0 (back to main menu of R4 NDSI by pressing START button)
DLDI auto-patching
Support Download play.

npcizzy, 2017-08-13:
I have that same card but it hails from r4ids.cn and it worked fine. Dumped and Injected with zero issues. (Shows up as "DORASU")

zoogie, 2017-08-13:
Unless the url on the sticker is www.r4ids.cn, it is not compatible.

Valeri0p, 2017-08-13:
This is a great example of description anyways ;)
@kitling perfect for an issue template...

cakehonolulu, 2017-08-13:
@zoogie I know, but i contacted @hedgeberg on Twitter and kindly asked me to make a flashcard's "Writeup" so they can look into it.

cakehonolulu, 2017-08-13:
@Valeri0p Many thanks! Aside from that, yes, it could be used as a template. It's not perfect, but i think it does the job for the developers as a first-sight to unsuported flash cards.

Dimensional, 2017-08-14:
I have to ask only because I have this, but given that the Blue Card for the Gateway is just an R4i Gold, would it be compatible too? It shows the exact same message when dumping the firmware, and it shows up as Deep Labyrinth on the Home Screen.

cakehonolulu, 2017-08-14:
@Dimensional No idea, wait for @hedgeberg 's and @kitling 's answer, it's better to not make "theories" about flashcards compatibility.

Alexdodala, 2017-08-14:
I have a similar problem I have a (not gold) R4i rts 3ds, bought from www.r4i-sdhc.com, which has the same chipID and HW REV but there is a few differences between this and the gold card should I submit the features and stuff if it can help? I'm not 100% sure if it is a genuine or fake/clone card but I want to help if I can.

kitling, 2017-08-14:
@Dimensional I've heard that it's an R4i gold card , I've heard that GW Blue is a cart that we can't actually update, so...

Only time will tell. Unfortunately, we don't know about everything.

Kubasinko, 2017-08-14:
Support for r4i-sdhc.com cards would be great.

Hikari-chin, 2017-08-14:
Please read the README guys, this is not the right place for this...

closed 2017-08-14 by Hikari-chin

hedgeberg, 2017-08-14:
Yeah @cakehonolulu the correct place for this is flashcart-core. We want to separate issues with cards and issues with this tool.