ntrteam/ntrboot_flasher #12: Problem injecting stay at 0%

Siogenon, 2017-08-13:
On acekard 2i H81, N3DS, latest release. Card is detected, dump work no problem. But when i inject, injection stay at 0%, same problem for restore, stay at 0%.

Tried latest files, but nothing.

Dimensional, 2017-08-14:
I have the same problem with my AK2i HW-44, one of the first bought within a year of it's release, so it only says compatible with DSi.

Siogenon, 2017-08-14:
I'm not only guy who have this problem o/.
Acekard2i dsi Compatible too ?

Juanito9001, 2017-08-14:
Your Acekard is with the last firmware and kernel?? sometimes that model of cards have problems for the firmware... or other times, the guilt is the micro SD card, because is a chinese model (false space and corrupted inside) try with another SD that you know is an original brand

Siogenon, 2017-08-14:
I tried with another SD ( 3). And i can't update firmware, don't have ds lite :/ my linker is not detected like game.

hedgeberg, 2017-08-14:
If you have a cracked 3ds, there's an app lying around somewhere that allows you to bypass any cart checks and initialize your cart. This may allow you to initialize the firmware update. Alternatively, give us the hw rev, and we can see if someone has an updated flash you can use. It may end up being, though, that your cart is broken and not useful.

Siogenon, 2017-08-14:
I tried with decrypt9 to flash my AK2I, but say me " cards is not AK2I".

Ntrboot detect my Ak2I like real ak2i h81 and can dump, but decryp9 nop.

hedgeberg, 2017-08-14:
mmm. @kitling does d9 do additional checks on the flash before doing stuff?

Siogenon, 2017-08-14:
I tried with ntr_luncher, but get white screen, maybe my ak2i is brick.

kitling, 2017-08-15:
@hedgeberg D9 also checks chipid, which I stopped doing to avoid softbricks due to overwritten chipid.

hedgeberg, 2017-08-15:
mmm. In that case, this is beginning to sound like an unwritable cart. Your cart may be hardbricked.

Siogenon, 2017-08-15:
Thank's for reply.

Yesterday, i tried, after a lot of Positioning, ntboor have injected on my linker. Maybe a bug, but 4 time injecte road 0% to 100%, but maybe a bug because my other N3DS bricked don't boot on.

I can't buy other cards, i have checked a lot of possibility for unbrick my ak2i with my ds onei mini/cyclo evo DS, but all need a NDS lite/phat, so i'm rekt.

So, we have possibility to see if ak2i is hardbricked ?

cakehonolulu, 2017-08-15:
So same problem as mine.
Any thing we can do?

Siogenon, 2017-08-15:
CakeHonolulu, ur flashcard work/not bricked ?

cakehonolulu, 2017-08-15:
@Siogenon My flashcard has the same problem as yours, cannot be written

Siogenon, 2017-08-15:
So, exactly same problem than me.

hedgeberg, 2017-08-22:
Sounds like the flash unlock was not successful, I don't really know how we'd go about debugging this remotely.

Siogenon, 2017-08-22:
I remain available if u need informations

Siogenon, 2017-08-25:
Latest release ( 0.3) crash, when i lunch from godmod9, 3DS just shut off

hedgeberg, 2017-08-25:
You mean the flasher crashes?

Siogenon, 2017-08-25:
Problem of crash solved ! ( need to update B9S). so, with the latest release, injecting stuck always at 0%

rootyElf, 2017-09-02:
I made [a D9 build without cart id check](https://www.dropbox.com/s/ikpdbxa3jathee9/Decrypt9WIP.firm?dl=1). Maybe you could give it a shot.

Siogenon, 2017-09-02:
RootyElf thank's, but i have only 2 choice  flash and dump.

And say me " Cartdrige is not AK2i".

rootyElf, 2017-09-02:
Ntrboothax options [have been removed](https://github.com/d0k3/Decrypt9WIP/commit/62c55f834cae069a5e7c836c73bbca0ea4b55e1f) because they are obsolete. Flash and dump are the only relevant ones anyway. As you have already found out D9 checks the cart in [this](https://github.com/d0k3/Decrypt9WIP/blob/855c63d/source/decryptor/ak2i.c) file. My build has the cartId condition removed. [This](https://www.dropbox.com/s/9lf10qm3zrxagbo/D9.firm?dl=1) build has the hwVer condition removed as well. With it there should be no way D9 complains about the cart not being an AK2i. You can try it, but it's obviously at you own risk.

Siogenon, 2017-09-02:
Thank's. so i try, i get " cart : 0X103e204f HW : ff.

Stuck on Erase flash.

Siogenon, 2017-09-07:
New Release say me just "flashcart setup failed ".

kitling, 2017-09-07:
Sounds like failed cart init. I don't really have any suggestions at this point, unfortunately.

Siogenon, 2017-09-08:
So, i think my linker is just destroy ? I'm waiting for ds onei compability, better close this issue maybe.

Thank's for help and for ur work.

mariomadproductions, 2017-09-16:
Same problem with my AceKard 2i HW44. Dumping and restoring work for me though.

kitling, 2017-09-28:
If you're concerned about a bad backup, go ahead and upload it and try using the proper backup from [here](https://3ds.guide/installing-boot9strap-(ntrboot)#section-v---removing-ntrboot).

dubbz82, 2017-09-30:
Will the recent additions for logging possibly help determine what went wrong with this?

kitling, 2017-09-30:
It's possible. Go ahead and send a log file. May build a debug build for you if there's nothing useful in there.

User13181408200115, 2017-10-15:
Same problem on DSTT also occurs with ds/dsi flasher does not occur with https://github.com/kitling/flashcart_core/files/1222726/ntrboot_flasher.zip but I am not sure if ntrboot is actually functional after flashing with https://github.com/kitling/flashcart_core/files/1222726/ntrboot_flasher.zip

HandsomeMatt, 2017-10-15:
> ntrboot_flasher.zip but I am not sure if ntrboot is actually functional
> flashcart_core/files/1222726/ntrboot_flasher.zip

DS/DSi flasher is probably a few versions behind which is why it's not
working. This isn't the issue tracker for that, this is the issue tracker
for the 3DS FIRM.

User13181408200115, 2017-10-17:
What I am saying is it fails using both a 3ds(.firm flasher) and with the nds flasher

BloodViper, 2017-11-16:
Any fix to this yet? I'm having the same problem on my DSTT

eltrolado, 2018-05-22:
Same problem with original DSTT... Any possible solution?