ntrteam/ntrboot_flasher #13: Acekard 2i HW81 not working

MrVtR, 2017-08-13:
The dump and inject work 100%,but when I put the acekard  2ds and made the combo,the screen doesn't change and the console boot on the main menu,someone can help with that?

zoogie, 2017-08-14:
Check the /boot9strap folder for boot9.bin, boot11.bin, or otp.bin. Are those files there?

For now, make sure luma/config.bin is deleted or luma3ds will just "pass through" after ntrboot. This is what happened to me.

MrVtR, 2017-08-14:
Nope,in /boot9strap folder I don"t have boot9.bin,boo11.bin or otp.bin. The SDcard doesn't have the /luma folder,because I didn't install any CFW on the 2ds. My New 3ds doesn't have CFW too,but the Exploit works,I tested in the new 3ds about one hour ago

zoogie, 2017-08-14:
Hate to ask about simple things, but you switched the 2ds into sleep mode, no?

MrVtR, 2017-08-14:
No,I read in another issue here to not switch the 2ds into the sleep mode,to only make the combo Start+Select+X

zoogie, 2017-08-14:
@MrVtR you need to have it in sleep mode. I don't know where you heard that misinformation.

MrVtR, 2017-08-14:
it works!!I got it,the Power button needs to switch in the same time of the sleep mode.LOL,noob error

closed 2017-08-14 by MrVtR