ntrteam/ntrboot_flasher #15: R4I GOLD (r4isdhc.com) support

Juanito9001, 2017-08-14:
[image]

Does this R4 card have support? 
i was reading the "flashcart_core" and it says "incomplete" (R4 SDHC Dual core) and the R4 of the image is from the family of that R4 in the "incomplete "section
so... the question is?

¿The R4 famili cards from "R4iSDHC.com" Will have support??

PD: Sorry for my english

hedgeberg, 2017-08-14:
It currently does not have support, but hopefully will eventually. If you have one of these cards, can you submit an issue on https://github.com/kitling/flashcart_core, with a picture of the actual cart, the reseller, and the hw rev/chip id read out from trying to dump using ntrboot_flasher?

Also, your english is fine, no need to apologize!

Juanito9001, 2017-08-14:
@hedgeberg thanks for the answer, i alredy submit the issue  in the "flashcart_core" space with the picture, the only things that i missed was the Chip ID and the HW REV, i will upload that info soon.

hedgeberg, 2017-08-14:
Don't forget reseller! we'd like to make an index of all carts and reliable resellers for those carts since there are so many counterfeits

Juanito9001, 2017-08-14:
http://www.nds-card.com/ProShow.asp?ProID=490 This is the page where i acquired the card, but i acquired the 2015 version.
good news : the 2015, 2016 and 2017 versions are exactly the same chip and everything (Kernel and firmware, I tested in a New 3ds with 11.4 and an Old 3ds with 11.5 and it works perfectly to playing NDS Games

hedgeberg, 2017-08-14:
Mmmmk, make sure to submit that information at flashcart_core, and we'll see what we can do when have a chance.

Hikari-chin, 2017-08-14:
Please read the README next time...

closed 2017-08-14 by Hikari-chin