ntrteam/ntrboot_flasher #16: More of a request than an issue

the-god-of-memes, 2017-08-14:
Would someone please support the original R4? It is very cheap (around $8) and lots of people have them.

astronautlevel2, 2017-08-14:
As far as I know, the original R4 is a ROM rather than NAND, which means it can't be reflashed, and as such can't be supported

the-god-of-memes, 2017-08-14:
In fact, it does.

ihaveamac, 2017-08-14:
got a link to an update it once had? this is the first I've heard of that (owner of original r4 here)

astronautlevel2, 2017-08-14:
We're talking about firmware updates, not menu updates.

the-god-of-memes, 2017-08-14:
http://www.r4wood.com/pages/wood-r4-kernel.html

ihaveamac, 2017-08-14:
that's the kernel you put on an SD card, not an update to the card itself.

the-god-of-memes, 2017-08-14:
Oh. Thanks for clearing it up.

the-god-of-memes, 2017-08-14:
Kind of off topic, but what is the cheapest flashcart that has cart updates?

hedgeberg, 2017-08-14:
That is currently supported? The r4ids.cn R4i 3ds RTS. Hopefully the SDHC or other cheap cards will be supported soon but as of yet they are not.

the-god-of-memes, 2017-08-14:
Alright, thanks.