ntrteam/ntrboot_flasher #17: Got an R4i variant that was detected as a compatible R4i Gold 3DS? Please post it here.

Hikari-chin, 2017-08-14:
There seem to be quite a few flashcarts that are merely clones or rebranded R4i Gold 3DS's, so we are looking to compile a list of compatible carts that fall under the R4i Gold 3DS umbrella. 

@kitling closed the other issue of this nature, #8, because of people reporting that their card did not work, asking when their cart would be supported, etc. So, let me be clear. This is only for reporting if you have an R4i variant that was detected as compatible and worked. Anything else will be deleted.

hedgeberg, 2017-08-14:
Also, for documentation's sake, if you post here, please include a link to the seller and a picture of the cart in addition to any other info.

dark-samus, 2017-08-15:
take your cart apart and post pictures as well, those help identify which cart is which pretty well.

Hikari-chin, 2017-08-15:
Additionally, pictures of the PCB inside the cart would be extremely helpful for identifying which particular variants as some cards have identical outer cases with different PCBs inside and thus have varying compatibility. However, you should not attempt to open your flash cart unless you know exactly what you are doing as you may damage it; we will not accept any liability if you do so.

HollMax, 2017-08-15:
r4idsn.cn is likely using the same PCB on all its products, as people are having succes with non RTS cards as well.

adiktus, 2017-08-15:
R4i Gold 3DS RTS (r4idsn variant)
Is this a typo? They come from r4ids right?

rootyElf, 2017-08-15:
The R4i Ultra (r4ultra.com) might be compatible as well since it seems to be a Acekard 2i clone. (https://gbatemp.net/threads/how-to-convert-an-r4i-ultra-r4ultra-com-clone-into-a-legit-acekard-2i-using-a-hacked-3ds.473764/)

Hikari-chin, 2017-08-16:
@HollMax Could you please post pictures of the supported carts, where they were bought from, and their PCBs?

Clector, 2017-08-25:
R4 3D Revolution from r4idsn.com Revision "C" or A7 works fine with all options, this one seems to use the same hardware as the r4ids.cn cart and restoring a dump from a matching revision cart from an r4ids.cn cart makes it works as one, gets detected as a r4ids.cn by their cart updater and updates fine. By the way the cart from "r4idsn.com" doesn"t seem to be produced anymore.
Here are some pictures of the cart and the PCB:
[image]
[image]
[image]

Anggeloko, 2017-10-15:
Hi, I have this r4 with this firm I could dump a backup but with the new firm it tells me that it is not supported, is there any option?

[image]

[image]

[backup3DSR4.zip](https://github.com/kitling/ntrboot_flasher/files/1385123/backup3DSR4.zip)

johnwraight, 2017-10-16:
R4i rts 3ds (www.r4i-sdhc.com)
https://imgur.com/a/VXYik
[image]
[image]
Confirmed working on the latest update!

zacchi4k, 2017-12-06:
R4i SDHC from website "R4iasdhc.com"

<details><summary>Images <em>(Click to Expand)</em></summary><p>
[image]
[image]
[image]
</p></details>>

Sadly my unit in particular broke, so I can't dump it or whatever

Bolt23, 2017-12-19:
https://encrypted-tbn3.gstatic.com/images?q=tbn:ANd9GcQTSYzlxaex4vwq8lRfeNEmXqcYiPUbFGIeRuBdYRG4CHJtBeY1dQ

I have this R4i v1.4 BLUE (No RTS or WiFi) ndsiLL.com which is detected as R4i GOLD 3DS, the injection successfully completes but I can still see the DS Cart on homemenu which clearly indicates it inset NTRBootHaxed! I cannot at the moment send you the pics of the Flashcart from inside out. Thanks in advance!