ntrteam/ntrboot_flasher #21: [QUESTION] Is the R4 Upgrade Gold Pro (2016) flashcard compatible?

GrenderG, 2017-08-15:
I'm not sure if this flashcard is compatible so I thought this would be a right place to ask. 😄 

[image]

Thanks in advance!

y0shimari0, 2017-08-15:
Just try it and report back if it works or not. AKA we dont know

GrenderG, 2017-08-15:
@y0shimari0 I can't try it atm, I opened this issue in case someone (apart from the devs) has tried it.

kitling, 2017-08-15:
What @y0shimari0 said. Just so you know, if it isn't compatible, this isn't the place to ask for someone to work on it. Read the README and follow the links, in that case.

It probably *isn't* compatible, but you never know.

lomejor18, 2017-08-15:
@kitling is it possible for you to make an app for flashing generic flashcarts? (for recovering purpose)

some time ago, i broke a chinese flashcart when tried to install YSmenu...
i would like to dump the flash from another flashcart (same model) and inject it on the broken one...
or that's nor possible?
thanks for your answer...

hedgeberg, 2017-08-15:
If we could do that we'd be able to support every flashcart already. We don't know the flash r/w commands, meaning we simply can't do that.

closed 2017-08-15 by hedgeberg

kitling, 2017-08-15:
@lomejor18 That's... kind of the point of getting everyone to help. This isn't generic yet, but with some time and patience, we can make this work on a lot of carts.

Hikari-chin, 2017-08-16:
Read the README next time.