ntrteam/ntrboot_flasher #23: Supported cartridge not detected : AK2i

WEctorr, 2017-08-16:
when trying do dump the flash, say:

ChipID: EA17C4FF
HW Rev: EA17C4FF
Supported cartridge not detected!

The decrypt9 show it as HW-81.

By ntr_boot flasher I can not do the dump, but by decryp9 yes.

WEctorr, 2017-08-16:
[image]

Hikari-chin, 2017-08-16:
Have you tried reseating the flash cart a few times? AK2i is known for contact issues.

WEctorr, 2017-08-16:
Yes, Unsuccessfully ;/

d3m3vilurr, 2017-08-21:
1. how about d9wip?
2. is this retail cart?

thisismike97, 2018-02-18:
@WEctorr Have you succeeded yet? Or can you retry using the latest version of Ntrboot_flasher?

I have a feeling that this issue could be closed.