ntrteam/ntrboot_flasher #24: I cannot restore Flash to my R4i Gold 3DS flashcart

yacsama, 2017-08-17:
Hi, I hacked my new 3ds LL successfully using the NDS flashcart R4i Gold 3ds.
When I try to restore it to its original state using ntrboot flasher 0.1.2 it says restoring complete but doesn't work anymore on NDSi, 2DS and new 3DS (not detected)
How can I fix it? or is it bricked?

hedgeberg, 2017-08-17:
It shouldn't be bricked, don't worry. Sounds like a failed flash, which should be fixable. Try redoing the restore flash step, see if it plays nice. If it doesn't, let me know, and we can walk you through restoring the whole thing.

yacsama, 2017-08-17:
Thanks, that was a fast reply. I tried restoring many times and still not detected.

hedgeberg, 2017-08-17:
Yeah you caught me at a good time, and I live to serve. It sounds like your backed-up flash may be h*cked, so we can work on having you restore that cleanly. Ill upload a clean flash somewhere  later. @kitlinh, what's needed to restore a full flash atm?

hedgeberg, 2017-08-17:
* @kitling

yacsama, 2017-08-17:
Alright, thank you man. I am using the gateway blue card on my NDSi now I am not in a hurry to fix it.

kitling, 2017-08-17:
In main.cpp, if you uncomment the `#define FULL_RESTORE`, then it will force a full write.

hedgeberg, 2017-08-17:
@yacsama can you take the time to build this? Let me know when you're done and I can send you the flash.

yacsama, 2017-08-19:
The problem is fixed by using new files from the updated 3ds.guide. I restored the flash using the new backup.bin extracted from R4i_Gold_3DS_RTS-Flashrom.zip. thanx

closed 2017-08-19 by yacsama