ntrteam/ntrboot_flasher #26: Not an issue but more of an idea on how to get ANY card's code stuff

QuoteStrife, 2017-08-18:
[image]

This is an idea i am putting in the ntrboot issue page as a way for people with actual skill to try out my idea.    Can we not use powerslaves to get into the card and reverse engineer it?

kitling, 2017-08-18:
That is one tool we have available to us, indeed. We still need *something* to give us an idea of how to read/erase/write flash.

hedgeberg, 2017-08-18:
I'm actually streaming working on this with the scds2+. Our process thus
far has been: dump the spi flash using flashrom on an rpi to get the flash,
look for any obvious code, then start reverse engineering any software
available to us, like firmware updaters, and then if all else fails we have
a flash backup and can write an app to fuzz the cart with magic numbers.

QuoteStrife, 2017-08-18:
oh well is there a way we can contact r4isdhc.com for that _something_

QuoteStrife, 2017-08-18:
also, did you guys try looking for an older model that may have something we can use and then try it on the r4i sdhc

Hikari-chin, 2017-08-18:
This is not an appropriate use of the issues page.

closed 2017-08-18 by Hikari-chin