ntrteam/ntrboot_flasher #29: Supported Cartridge Not Detected: R4i Gold RTS Wi-Fi

wolfpacleader1986, 2017-08-24:
Have a N3DS on custom firmware. Ran the Acekard Flasher which is supposed to work with the card(the nds file), and I get a message that the cartridge is not supported. Just bought it a few days ago and it's a legit cart.

hedgeberg, 2017-08-24:
This doesn't make any sense, what do you mean the nds file? This tool doesn't use nds files. What cart? You said you bought it a few days ago but how do you know it's "legit"? Where did you buy it? We need a lot more information. Please provide the cart's Chip ID and hw rev as listed when trying to dump using this tool, as well as a link to where the cart was purchased and if possible pictures of the circuit board inside the cart's plastic.

wolfpacleader1986, 2017-08-24:
Bought it from dsflashcart.com. I know it's legit because it had the authenticity holofoil sticker on the back. It's the R4i Gold RTS Wi-Fi. Chip ID is 000007C2. HW Rev is FFFFFFFF.

Originally used the NDS file for the single system method, but it doesn't work on your custom firmware tool either.

Juanito9001, 2017-08-24:
Can you upload photos?? Specially of the PCB

_Enviado desde mi Samsung SM-G610M usando [FastHub](https://play.google.com/store/apps/details?id=com.fastaccess.github)_

wolfpacleader1986, 2017-08-24:
I also forgot to mention, the code comes up as legitmate.

wolfpacleader1986, 2017-08-24:
What file host for the pictures?

kitling, 2017-08-24:
The cart with the label 'Wi-Fi' is not currently supported. See https://github.com/kitling/flashcart_core/issues/14

closed 2017-08-24 by kitling