ntrteam/ntrboot_flasher #33: Problem restoring flashcart

Juanito9001, 2017-08-27:
I do the ntrboot to a flashcart r4igold wood, it works, but when i restored, the 3ds or a nds doesn't recognize it, in the case of the 3ds, it freezes if i insert it or if i turn on the 3ds with the flashcart inside, whats going on? 
And another question... the flashcart still works for magnethax if i inject the ntrboot again? Even with the error i mentioned? 

_Enviado desde mi Samsung SM-G610M usando [FastHub](https://play.google.com/store/apps/details?id=com.fastaccess.github)_

Anggeloko, 2017-09-05:
@Juanito9001 use this backup and tell me if it works again the flashcard

[backup.zip](https://github.com/kitling/ntrboot_flasher/files/1278683/backup.zip)

Juanito9001, 2017-10-19:
I tried with the actual version of the ntrboot AND another backup and finally works...
using the same backup but with another version doesn't work...
anyway... thanks

closed 2017-10-19 by Juanito9001