ntrteam/ntrboot_flasher #38: DSTT dumping but unable to access luma through ntrboot

Rainoutt, 2017-09-04:
Hello, i was able to compile ntrboot_flasher.firm. With it I dumped and even flashed the dstt flashcard with chip MX29LV400B (supported), but when I tried to boot the luma nightly version, it told me it booted from SD via B9S, even when I did the right button combo and my Old 2DS was on sleep mode, I am doing something wrong to check if the card is working? Thank you for your work, I'll post images or do anything to help.

HandsomeMatt, 2017-09-05:
What are you using for `boot9strap_ntr.firm` and how big is the file?

Rainoutt, 2017-09-05:
I was using the sciresM 1.2 ntr version, should I also compile it? the file is 68,6 kB (68608 bytes)

HandsomeMatt, 2017-09-05:
The DSTT only has 64KB of flash memory. The newer version of b9s is a 30KB
FIRM that will fit, you will have to compile yourself however as no release
has been made yet.

HandsomeMatt, 2017-09-05:
In the future ntrboot_flasher will check to see if the FIRM can fit in the
flash

Rainoutt, 2017-09-05:
I compiled b9s and flash it, but luma3ds nightly still shows "booted from SD via B9S", is this the intended or there is another way to test if the flash worked?

closed 2017-09-05 by Hikari-chin