ntrteam/ntrboot_flasher #41: Can't restore Acekard 2i's flash to R4i Ultra

s1cp, 2017-09-06:
Everytime I try to restore an ak2i flash to my R4i Ultra (r4ultra.com) ([direct ak2i clone made by team acekard with the same hardware](https://gbatemp.net/threads/how-to-convert-an-r4i-ultra-r4ultra-com-clone-into-a-legit-acekard-2i-using-a-hacked-3ds.473764/)) it gets stuck at the same spot, as shown below:

[image]

I am able to restore an ak2i flash to the r4i ultra using Decrypt9 just fine, but not with ntbroot_flasher.

Hikari-chin, 2017-09-06:
Could you try the latest release? Getting stuck at 49% was a general AK2i problem in some more recent post-0.1.2 but pre-0.2 commits.

s1cp, 2017-09-06:
Tried the new 0.2 version, still gets stuck at the same spot.

s1cp, 2017-10-16:
I tried it again, but now with ntrboot_flasher 3.1 and with loglevel switched to debug.
The top screen got stuck at "writing chunk 001F0000" while the bottom screen had a progress bar which was moving slowly until it got stuck at 50% and I had to do a force shutdown.
Here's the log file:
[ntrboot.log](https://github.com/kitling/ntrboot_flasher/files/1387437/ntrboot.log)

s1cp, 2017-10-16:
Some PCB pictures:

[image]
[image]
[image]
[image]

MushGuyDS, 2023-05-13:
Same problem here, but instead it kept getting stuck at 25% with latest release.