ntrteam/ntrboot_flasher #42: [DSTT] M29W400B is compatible, datasheets is the same as M29W400T

annson24, 2017-09-06:
3 DSTTs I have have flashchips of ST M29W400B. I've successfully injected ntrboot on 2 out of 3 of them. The one that didn't worked had hardware problems anyway since it was really old and was collecting dust; even after cleaning it's still going haywire.

Anyway, checking online, seems like M29W400B and M29W400T has the same datasheet yet M29W400T was listed in the unsupported section.

http://pdf.datasheetcatalog.com/datasheet/SGSThomsonMicroelectronics/mXttvuu.pdf

HandsomeMatt, 2017-09-06:
You're correct, a lot of the "unsupported" carts are from another program called DsttFlashWriter. I didn't have the time to check every single flash chip.

I've committed locally to add support for that chip and a few others, it'll be pushed out later today.

Can you post in flashcart_core in the future for flashcart specific stuff, thank you.

closed 2017-09-06 by HandsomeMatt

annson24, 2017-09-06:
Yes sir, just saw your thread there. Will post there after testing more chips.