ntrteam/ntrboot_flasher #43: MX29LV320 doesn't work.

mark201200, 2017-09-06:
I have a DSTT clone, it uses the MX29LV320CBTG-90G chip.
I've checked the data sheets and it should be supported, but when I try to flash NTRBoot it says Flashcart setup failed.

Shinigati, 2017-09-06:
I don't know if this will help but it did help me with my ak2i clone, if you have access to cfw or a nds/dsi try using the dsi flasher rom first. My ak2i was detected as not compatible in v 1.3 until i flashed it using the rom then it would let me dump inject or restore.

mark201200, 2017-09-07:
Can someone compile a version of the installer that treats every card as a DSTT with a MX29LV320 ( ID: 0xA7C2 ) chip?

HandsomeMatt, 2017-09-09:
Please keep flashcart specific stuff in https://github.com/kitling/flashcart_core/

closed 2017-09-09 by HandsomeMatt