ntrteam/ntrboot_flasher #45: Regression: ntrboot_flasher no longer reports the ChipID and HW ID

OronDF343, 2017-09-07:
Title says it all. Previous versions reported this info which is useful for obvious reasons.

kitling, 2017-09-07:
ChipID is almost never useful, hwid only in certain situations.

Handsomematt is planning on making a debug menu with this kind of stuff, if I read correctly.

HandsomeMatt, 2017-09-09:
ChipID is completely useless and always returns FC2 and HWrev was only specific to AK2i clones iirc.

I think my plan is to just have Ak2i carts/clones return their hwrev when unsupported and maybe some sort of debug menu that prints: header, chipid and maybe the hwrev command if it works.

Valeri0p, 2017-09-09:
Some cards reported fc7/fb7 as chip id, so maybe isn't that useless...

kitling, 2017-09-28:
@Valeri0p It really doesn't make any difference to us.

Regardless, logging is now implemented and v0.3.0 has it! HW Revision is now printed to the log, meaning that we can now close this issue.

closed 2017-09-28 by kitling