ntrteam/ntrboot_flasher #53: cloning with --recursive is broken

urherenow, 2017-09-29:
is it really necessary to reference a specific commit in the first place. Just include the dependencies until/unless a commit to them actually breaks something here.

HandsomeMatt, 2017-09-29:
what are you even saying, we use a submodule because it makes way more sense

urherenow, 2017-09-29:
You see "flashcart_core @ 6d9e3b6 " ? Lose the @6d9e3b6
same for libelm3ds.

Otherwise you get crap like this when commits are reverted `Cloning into 'D:/Projects/ntrboot_flasher/libelm3ds'...
remote: Counting objects: 44, done.
remote: Total 44 (delta 0), reused 0 (delta 0), pack-reused 44
error: no such remote ref 6d9e3b6ba69f7a83cf0863234429e26c6807775a
Fetched in submodule path 'flashcart_core', but it did not contain 6d9e3b6ba69f7a83cf0863234429e26c6807775a. Direct fetching of that commit failed.`

as I said, pointing to a specific commit is never necessary, unless a future update to a submodule breaks something you are using it for. 

Frankly, I'm convinced you are an idiot for your reply. I never said not to use submodules.

ihaveamac, 2017-09-29:
that's called a submodule. you have to point to a specific commit. that's how submodules work.

Hikari-chin, 2017-09-29:
Closed, and I edited your comment to be less hostile. Don't act this way, please.

closed 2017-09-29 by Hikari-chin

kitling, 2017-09-29:
@ihaveamac There probably is a feature where it just grabs the latest.

@urherenow I'm not the most familiar with git submodules. It's setup to reference specific commits because I didn't know you could do it without that.

ihaveamac, 2017-09-29:
if there is a feature to always get the latest, I don't know about it. I only see it supporting pointing to a specific commit, and guides on how to update it.

urherenow, 2017-09-29:
"git submodule add https://github.com/kitling/flashcart_core flashcart_core" 

on a command line/terminal tags on a specific commit for when you try to clone rercursively?

ihaveamac, 2017-09-29:
this will add it as a submodule at the latest commit at the time of adding. it still doesn't automatically pull new changes.

urherenow, 2017-09-29:
ok. But it needs to be done. As I said a clone --recursive is currently broken. 

"git submodule update --init --recursive" before a push should do the trick, no?

urherenow, 2017-09-29:
If you're using git 1.8.2 or netter, "git submodule update --remote --merge" will do the trick before a commit.

angelsl, 2017-09-29:
@urherenow

It does not make sense to have Git checkout the head of a branch instead of a specific commit for a submodule. Each commit tracks a specific state of the entire repository that should remain identical across time, including submodules, so that if you were to checkout a specific commit you would get the same tree regardless of when. Kind of like a blockchain.

In fact you mentioned it yourself:

> unless a future update to a submodule breaks something you are using it for

This is practically guaranteed as code is refactored, moved around, etc. It breaks the entire model of Git if things tracked _in the tree in a specific commit_ change without the commit hash or tree hash changing. Kind of like a blockchain.

If the submodule reference is broken, you can update it and make a pull request. No need to be so belligerent.

urherenow, 2017-09-29:
@angelsl wrong. first tell my how THIS repository got messed up then. Then, read the links I posted.

kitling, 2017-09-29:
Sooooo... this whole thing boils down to 'do the same thing you're already trying to do but failed at once'. Gee. How helpful. For the record, I've already read the documentation on submodules, but that just means I've absorbed the basics, which means I try to leave doubt for things I don't know. Next time, could you please not say "why do you not do this thing with your tools" before you find out if it's even possible?

It broke in the first place because I do my active flashcart_core development in the submodule tree to keep it convienent for testing before committing. I specifically remember trying to update the flashcart_core commit being used before merging the update stuff into ntrboot_flasher, but I guess I failed.

I'll fix this tomorrow at some point when I'm on my laptop.