ntrteam/ntrboot_flasher #56: Original R4i 3DS rts not detected.

Cyrox49, 2017-10-01:
Hi,

My R4i is brick.

I would like to use ntrboot_flasher to flash my backup.bin of linker but card is not detected.

Can you help me pls ?
[image]
[image]
[image]

HandsomeMatt, 2017-10-01:
This is not the latest version of ntrboot_flasher. https://github.com/kitling/ntrboot_flasher/releases Try again.

Cyrox49, 2017-10-01:
Thanks but same problem :/
[image]

Cyrox49, 2017-10-01:
I have try previous version too of ntrboot and i have the message : "flashcart setup failed"

I don't know how to recognized my linker by ntrboot flash

kitling, 2017-10-07:
Reopening until someone has the time to check whether this is actually a supported cart that isn't being detected or not.

Cyrox49, 2017-10-08:
The chip ID and HW rev of my linker is strange. Maybe a hardware brick but i'm not sure

Exayl342, 2022-06-27:
[image]
Same problem

Exayl342, 2022-06-27:
I have the same problem I'm using latest ntrboot flasher :( 
[image]
[image]

Exayl342, 2022-07-09:
> Thanks but same problem :/ [image]

> 

Have you fixed it?

ash20493, 2022-07-10:
Ntrboot flasher is not currently in active development, they haven't updated it since 2017, it gives the reasons on the readme.md, your best bet if you cant find a cart it will work with is to buy a premade cart or a cart that has an ntrboot function, they are relatively cheap these days, i know it's not ideal for everyone, but unfortunately there are few options for ntrboot hax, have you tried seedminer? That is the latest method for hacking the 3ds

Exayl342, 2022-07-21:
Btw i fix it :)

kitlith, 2022-07-21:
@Exayl342 If at all possible, in case other people have the same issue, could you describe what you did to fix it?

Exayl342, 2022-07-30:
@kitlith I use the ntrboot_flasher.firm v0.4.0