ntrteam/ntrboot_flasher #58: My DSTT doesn´t work properly at last release

IvanMolina99, 2017-10-02:
Hi! I have a problem with last Ntrboot-flasher version.
 I bought some weeks ago a DSTT and i received it today, but when i tried to flash my flashcart with the latest release of Ntrboot-flasher, it doesn´t works :(
It finished properly the backup but when i try to inject, it stop at 99% reading
         
[image]

The really reason of that issue is that i tried with the older version, the 0.2.0 of ntrboot-flasher and it works properly and inject the NtrBoot exploit succesfully. 

What can i do?

kitling, 2017-10-02:
Alright, so obviously there's been a regression. Thank you for bringing it to our attention.

Could you run this debug build and give us the logfile at /ntrboot/ntrboot.log? https://up1.secretalgorithm.com/#XWVMTG6EtX3nCUZLHKG1qA

Valkry, 2017-10-03:
Hi, i have the same problem as IvanMolina99 described before, the only difference is that my dstt was not supported on older versions, i have ran the debug build but it not produce the log file. I hope that this help to improve this amazing tool. By the way my chip is a 0x9389: INTEL 28F800B3B. Thanks in advance.

kitling, 2017-10-03:
@Valkry It's not the same issue. Theirs is a regression and belongs here, while yours is not and belongs at this issue on the flashcart_core repos: https://github.com/kitling/flashcart_core/issues/56.

Also, a better debug build (since appearantly fflush doesn't work?) here: https://up1.secretalgorithm.com/#3Ai8X-28PPWGVysaMqmDUg

roflpwnt, 2017-10-04:
I also hang at 99% verify on latest release but it works on prior releases. DSTT

[image]

kitling, 2017-10-04:
@roflpwnt Could you please run the debug build linked in the post right above yours? Thank you.

IvanMolina99, 2017-10-04:
Thanks @kitling I tried with the debug build that you linked before and it works fine 👍

kitling, 2017-10-04:
@IvanMolina99 wait wait wait... so the debug build *fixed* the issue?

IvanMolina99, 2017-10-04:
@kitling Yes, the release doesn´t works, it stopped at 99% reading but this build works, it inject Ntrboot and it's functional.
The problem is that it takes around 10 minutes in write the flashcart but it fixed the problem perfectly :)

kitling, 2017-10-04:
That's the issue though. It's not supposed to be a fix.

I guess it's some timing issue, so taking the extra time to open and close the file every time does it... urgh.

Tomorrow I'll send you a build with fopen+fclose but w/o the extra logging.

TLuigi003, 2017-10-04:
I've had the same problem(and 0.2 also fixes my issue). I think is not relevant but just if you need the info my chipset is Macronix MX29LV400B
Thanks for the work BTW.

roflpwnt, 2017-10-04:
Log here running the debug build : https://up1.secretalgorithm.com/#D_5PhTotSId8Pzz3TA0KVQ

kitling, 2017-10-04:
@roflpwnt looks like it's getting stuck at the same point here: https://github.com/kitling/flashcart_core/issues/56#issuecomment-334000898, but for a different reason.

@TLuigi003 use this build to provide a log file: https://up1.secretalgorithm.com/#3Ai8X-28PPWGVysaMqmDUg

TLuigi003, 2017-10-04:
@kitling Log with the debug build: https://up1.secretalgorithm.com/#ThyHu2hEcd4JB0Jhlo7LDg

The debug build fixed the issue although it took forever to complete the write section(probably because the log?), just as OP stated

kitling, 2017-10-07:
Try this build: https://up1.secretalgorithm.com/#xgeexFwlDf0M3WFKXGCZDg

It isn't a debug build, it's just something that I noticed that might have caused the regression.

TLuigi003, 2017-10-07:
This last build doesn't seem to fix the issue. It still stuck at 99% reading while injecting.
I've also discovered that it also get stuck if it tries to restore the flash to the modified flashcart(it works ok trying to restore and already restored cart)

I attach a photo here if it helps(sry for bad quality)

[image]

HWNJ, 2017-10-07:
I made a backup before the testings. After testing "2484-dirty", I made a new backup because it said "Erasing Flash" on the log.
Both backups are equals (CRC32: 48161804).
For some reason 6e9d-dirty saves a 0kb log. 2484-dirty instead gives a full log.
I puloaded here [some logs, info, pics and a videos from 6e9d-dirty and 2484-dirty](https://github.com/kitling/ntrboot_flasher/files/1366234/ntrboot.zip).
Here is the [same file with no video](https://github.com/kitling/ntrboot_flasher/files/1366245/ntrboot-novideo.zip) just in case.
I also uploaded [the video on youtube](https://youtu.be/arVoxewZTGI)

kitling, 2017-10-07:
There have been some issues with log file saving when stuff freezes. And when I create a build that sucessfully saves the log, then suddenly stuff is magically fixed for some people. which, of course, isn't a proper fix.

@HWNJ Did your card work on the previous release?

kitling, 2017-10-07:
Regression should be fixed by https://github.com/kitling/flashcart_core/pull/73, will leave this issue open until a release with the fix is published.

roflpwnt, 2017-10-08:
> Try this build: https://up1.secretalgorithm.com/#xgeexFwlDf0M3WFKXGCZDg
> 
> It isn't a debug build, it's just something that I noticed that might have caused the regression.

Hangs at "Reading 99%"

HWNJ, 2017-10-08:
@kitling 
My card do not worked on 0.2.0
That's why I stated it is a 9089 INTEL 28F160B3T.
I hope you can do something with the dump I provided you and with the log file

by the way, did you have a compiled https://github.com/kitling/flashcart_core/pull/73 version?

kitling, 2017-10-08:
@HWNJ this issue here was specifically about carts that worked on v0.2.0 but not on v0.3.0. General "my cart never worked please make sure it works" goes here: https://github.com/kitling/flashcart_core/issues/56

Someone also compiled a build and put it in that issue. I don't think it'll fix your issue, though.

HWNJ, 2017-10-11:
Well I will test that build, anyway, I hope my DSTT dump helps you.
Will report here the results of my tests tomorrow.

HWNJ, 2017-10-11:
The new build does not work...
It freezes at 0% after reading

HandsomeMatt, 2017-10-11:
Can you provide the log

HWNJ, 2017-10-11:
I do not have it right now, but I can upload it tomorrow at noon.
I am far from home right now.
It was a 2.57MB log file on debug mode
I remember it says ERASING then it sends some commands with some hex numbers and then started to send just 00 00 00 00 00 00 commands

HWNJ, 2017-10-11:
and if it helps, this is my [backup file](https://github.com/kitling/ntrboot_flasher/files/1364686/DSTT.-.9089.-.INTEL.28F160B3T.zip), I uploaded it some days ago

kitling, 2017-10-14:
Fixed with v0.3.1

closed 2017-10-14 by kitling