ntrteam/ntrboot_flasher #61: Error with DSTT 9089

HWNJ, 2017-10-07:
Just like happen to the user on issue 58 ( https://github.com/kitling/ntrboot_flasher/issues/58 ), my DSTT stops at 99% reading.

I see it is on the flashcarts that should probably work, but you have no specific datasheet for it.
I make a dump of my card's firmware:
http://www98.zippyshare.com/v/cXAbq8Oj/file.html
I hope it helps you to give full compatibility with this card. As I said, it is a DSTT - 9089 - INTEL 28F160B3T

HWNJ, 2017-10-07:
The debug builds did not worked.
I post some reports on issue https://github.com/kitling/ntrboot_flasher/issues/58 
Anyway, here's a copy of them:
[Logs, info, pics and a videos from 6e9d-dirty and 2484-dirty](https://github.com/kitling/ntrboot_flasher/files/1366234/ntrboot.zip)

kitling, 2017-10-14:
Actually belongs in https://github.com/kitling/flashcart_core/issues/56, where a bunch of other people are having the same issue.

closed 2017-10-14 by kitling