ntrteam/ntrboot_flasher #64: r4i sdhc not working

r4isdhcnotworking, 2017-10-14:
The new update says its working but it wont it says
Dumping flash
ChipID: 000007C2
HW Rev: FFFFFFFF
Supported cartridge not detected!

kitling, 2017-10-15:
could you provide images of your cart?

I suspect that they had several different carts, just the ones that we kept dealing with were all the same.

Clydefrosch22, 2017-10-15:
i'm just gonna hop in on this one.
the r4i-sdhc.com 4.4 revision doesn't work. the flasher just tells me its not compatible

image:
https://imgur.com/a/pzxZa
sorry for the shoddy quality

kuwoh, 2017-10-15:
flashcart setup failed: R4i 3DS RTS

mariogamer2, 2017-10-15:
Same issue with my r4i sdhc 3ds wifi.

@Eddy119 posted a flasher in the flashcard core issue. Maybe this could be used.

jihao, 2017-10-16:
Same issue with my R4 GOLD Pro RTS 2017 from r4isdhc.com

bbheko25, 2017-10-19:
R4isdhc 2013  cart not compatible

bbheko25, 2017-10-19:
R4 2013 https://imgur.com/gallery/F3o18

Emiliano1502, 2017-10-19:
Same issue with my R4 sdhc dual core

CaelThunderwing, 2017-10-20:
tried compiling the latest commit as of 10 hrs ago,  went to try flashing my R4 SDHC Dual Core (2015) it's going Abysmally super slow in just "Reading NOR" (2% in just 2hrs >_>looks the same as the same "2017" version same card color label except "2015" for the year on it, intead of 2017.

bbheko25, 2017-10-20:
Is there a guide how to compile?

Bestfast, 2017-10-24:
@bbheko25 it's easy. Just past this in your terminale: git clone --recursive https://github.com/kitling/ntrboot_flasher/ && cd ntrboot_flasher && make . Obviusly you must have "git" installed in your system. (sorry for my bad english, i'm italian)

maniack11, 2017-11-28:
Huh. which sdhc r4 are you using, rts or wifi?

bbheko25, 2017-12-02:
Dualcore 2013

kitling, 2017-12-18:
The dualcore 2013 cart is actually a different cart, iirc more closely related to r4igold.cc and r4isdhc.hk. Apologies for not responding to this sooner.

Squirrel61, 2018-08-03:
Is there any chance that the R4i SDHC 2013 (theoretically) can be flashed with ntrboot? It seems there are a lot of them still being used. Further, the "lite" is identical and the "new" (I think this is actually the 2012 version) is _almost_ identical, making for a really huge number of possible ntrboot cards.

But I can imagine that hardware-wise, it's impossible that the card ever can be converted. Can someone with more knowledge share a bit of light on this?

kitlith, 2018-08-04:
So, it's likely possible. Is anybody working on it? Unfortunately not. We've all been busy with other stuff.

Squirrel61, 2018-08-04:
Thanks for the info. I guess the software that needs to be put on the card is pretty much the same as for other R4i SDHC cards, so what is the missing link? Reversing of the original flasher, to get the flashing process?

kitlith, 2018-08-04:
See a bit earlier, "