ntrteam/ntrboot_flasher #68: r4 sdhc dual core 2016 ChipID: 00000FC2 HW Rev: 2A2A2A2A

kanogunz0789, 2017-10-15:
i deleted ntr boot dump files on my pc after doing the hack and i downloaded ntr boot to restore my r4 however it showed "flash not found' 
can you upload r4 sdhc flash files so i can restore my r4 !

kitling, 2017-10-15:
https://gbatemp.net/threads/r4i-b9s-flashcart-tested-and-impression.484620/page-18#post-7630897

This thread has a lot of dumps. Try them.

closed 2017-10-15 by kitling

kanogunz0789, 2017-10-15:
kitling i found just r4 sdhc rts no dual core i restore my r4 sdhc dual to rts bakup and is working fake card problem now how i found r4sdhc dual core bakup

kitling, 2017-10-15:
Look farther in the thread? I don't have an exact answer for you as I generally don't collect backups. Sorry.