ntrteam/ntrboot_flasher #72: Backup Process with R4iSDHC Family is painstakingly SLOW. / AK2i Restore Broken

CaelThunderwing, 2017-10-20:
as my AK2i finally craped the Bed (more below), i decided to use my R4 SDHC Dual Core "2015" as my backup NTRBoot card, and i had no idea if there was a backup for the "2015" card, so i compiled the latest commit as of Yesterday.  (i had to make a copy of the blowfish_ntr.bin via Ram dump taken from GM9 so maybe this Attributed to the Issues.)

For the AK2i Issue: i didnt know if compiling would of gone Awol/failed if i incorrectly created the required NTR bin from teh ram dump containing the blowfish data for DS Carts, but when i tried to test by restoring a backup of the AK2i it "Endlessly Looped" as if i removed the card or it failed during flash. (with no log produced) w/ not even a full restore from the last B9S version of Decrypt9 restoring the card (this is the only reason i keep that .firm on my sd card)

for the R4 SDHC Dual Core 2015 card issue, the whole process took over 12 hours to Dump the NOR and left in its wake (i set log level to debug) a 67MB Log file. a test Restore completed in less than a full min and cart went back to Flashcart functionality. i attached the build i made 'n used along w/ a compressed copy of the log. hopefully any of it helps finding out why the NOR Dump took so abnormally long! 
[ntrboot.zip](https://github.com/kitling/ntrboot_flasher/files/1403086/ntrboot.zip)
[ntrboot_flasher.zip](https://github.com/kitling/ntrboot_flasher/files/1403089/ntrboot_flasher.zip)

HackerBoy0412, 2017-10-21:
can say how you compile the last commits

HackerBoy0412, 2017-10-23:
can you comile the latest commit

angelsl, 2017-10-23:
> why the NOR Dump took so abnormally long!

The NOR dump took so long because you turned on debug logging, which means for every 4 bytes read from the NOR, 3 lines are written to the log, and for each line (for reliability reasons) we open and close the file.

Please don't turn on debug logging unless you are actually debugging.

dark-samus, 2017-10-28:
Sounds like PEBKAC issues. Closing.

closed 2017-10-28 by dark-samus