ntrteam/ntrboot_flasher #78: Possible to edit backup.bin?

kuwoh, 2017-12-18:
Is there any way to edit the backup.bin?

hedgeberg, 2017-12-18:
...what? Why on Earth would you want to do that?

kuwoh, 2017-12-18:
somehow to edit rom content? or patch stuff inside, etc.

hedgeberg, 2017-12-18:
What would that gain you? How would you benefit by modifying the rom contents? I'm not trying to be cagey but what you're talking about seems to be without motivation and could be potentially disastrous if done incorrectly.

kuwoh, 2017-12-18:
Well it's pretty much to put the auto boot flag to my flash cart, I've tried hex editing the contents of an arm7.bin, arm9.bin, header.bin, etc. yet those didn't work cause i'm probably missing something. My best bet to make such a thing work is through an editor

hedgeberg, 2017-12-18:
Well, frankly I don't know that much about rom editing, but it sounds like
you may be forgetting to update the signatures/checksums over the headers.
I don't know of any tools that do what you're talking about specifically
but this feels like a lot of work for minimal reward and I'd suggest not
pursuing it. Best recommendation I can make is reviewing the rom structure
on gbatek. However, this is not an issue with ntrboot_flasher nor
flashcart_core, so I'm likely going to close this issue.

HandsomeMatt, 2017-12-18:
In most cases your backup.bin is an NDS file, edit it with ndstool if you
must, but at your own risk.

kuwoh, 2017-12-18:
i think i tried editing it with ndstool before and it returned a much larger size if i recall correctly

HandsomeMatt, 2017-12-18:
Idk man, this stuff is advanced and it seems you don't really know what
you're doing.

kitling, 2017-12-18:
uh, backup.bin is not a NDS file afaik. It's a raw flash backup that varies depending on the cart backed up (one TODO is to have a different name for each cart). You *can* modify and reflash it, we do no checks of our own on it. Be careful, however -- I'd reccomend having a physical programmer to read/write flash in case something goes wrong. We will not support restoration of images other than original images or replacement images downloaded in the event of a missing backup, though.

closed 2017-12-18 by kitling