ntrteam/ntrboot_flasher #82: New Release should be uploaded

Bolt23, 2018-01-14:
As there have been many commits since latest release and you have updated the Flashcart Core, don’t you think it will be a nice idea to release 0.3.2? I mean many unsupported Flashcart are now supported!

Bolt23, 2018-01-24:
Well here it is 
[ntrboot_flasherARM.zip](https://github.com/ntrteam/ntrboot_flasher/files/1659731/ntrboot_flasherARM.zip) v0.3.2 (I compiled it :D 🥇 )

kitling, 2018-01-25:
okay, so, look. there's stuff that we're in the middle of. We may release a pre-release or something soon, but I don't feel comfortable with making a full release atm.

Bolt23, 2018-01-27:
@kitling Okay mate! As you say, but do you have any update on my R4i SDHC (ndsiLL.com) BLUE (colour) 1.4? I get the norErase4k error, though I submitted the log their at flashcart_core issues. I will be highly obliged and honoured too if you help me out with that..... :D

closed 2018-01-27 by Bolt23

kitling, 2018-01-29:
No, I have not ignored/muted you. I just haven't been paying much attention over here in general.

The flashcart issue should stay open. Just because nobody is actively working on it doesn't mean the cart doesn't exist.