ntrteam/ntrboot_flasher #83: R4i sdhc v2.0 happybox Red version v1.4

serpinu, 2018-01-18:
Hi, as the title my r4 worked but then Iwanted to try to make it work on 3ds and did not go then I asked on various forums and they told me to put a kernel and the firmware of the obsolete 3ds version (type version 4) and l 'I started and remained on loading, thinking it was blocked, I turned off the ds lite. Today I tried again and appeared in place of the icon of a star wars game appears top spin 3 2k. What can I do? I read that you have to do the flash with another flashcard but I only have a ds-xtreme. Thank you so much and sorry for my bad english

Bolt23, 2018-01-18:
Is your 3DS Hacked?
Does it have noob-proof CFW Luma3DS? (Or any other with ds cart block list patch)
It isn’t bricked if you didn’t flash something other than your backup on it, no kernel or firmware or TTMENU or YSMENU can brick it..... Though I heard some original kernels released by DSTT or maybe R4i SDHC used to brick the FAKE carts, if you used the originals then ‘maybe’ you’re cart is DEAD!

Don’t be Sorry for your English :D but try to improve it, cuz what you wrote was barely understandable. :P

serpinu, 2018-01-18:
Ahaha ok thanks, Yes my 3ds have luma 9.0 with last fw 11.6, the files i took from here https://github.com/ntrteam/flashcart_core/issues/111
And i put the kernel and the firmware of 3ds in the ad because on a forum told me ti do that. But now when i try to start the r4 in ds lite appears the Black screen, on 3ds the White screen...

Bolt23, 2018-01-19:
@serpinu  The original beta release that you used from R4i SDHC 'may' have bricked your cart, but still you didn't answer did you flash something or not? If it did work before use the kernel you used before. :D

serpinu, 2018-01-19:
@Bolt23 i used the file in the rar of that link i post , when I inserted the card in the ds it stopped on the writing loading, and I thought it was stuck on the screen I turned off the ds. Before this it was fine that I wanted to use it on 3ds, in fact It did not see the 3ds before. after that the icon has changed from the game of star wars to top spin 3 2k, something happened.
When i insert the original kernel It start but With a Black screen. Now the 3ds sees the card bit when i start the card, It start with a White screen, only difference with the ds.

Bolt23, 2018-01-19:
@serpinu Your card is bricked because the the kernel update must have triggered something to brick it...
 The icon change says that it was flashed by the kernel that make it 99% DEAD unless some dev can provide with you a backup flash for your (Pssst..... If you flash the original ROM your fake cart will become a LEGIT R4i flashcart) Who knows what can happen..... Wait for a dev support.

Though if you don't want to ntrboothax then why open issue here? Its not for cart time bomb and brick fixing sort of things....

serpinu, 2018-01-19:
ok, I'll wait, thanks. because I did not know where to open it nobody knows about this problem ... I also tried with ntrboot flasher but says that my card is not supported. I have to wait for the developers