ntrteam/ntrboot_flasher #84: How to compile

Siul89, 2018-01-19:
Hello, can someone tell me the steps to compile it?. I think I have everything installed but I have no idea what I have to do next.

Thank you

HandsomeMatt, 2018-01-19:
Run `make`

Bolt23, 2018-01-19:
@Siul89 Can you upload the latest commit that you compile for me too? It will be a big help, as I do t have a ready environment and my MCU is under repair and I wont be able to install those scripts on my MacBook Pro (I ain't that developer type ;D)

00junn, 2018-03-10:
https://raw.githubusercontent.com/ntrteam/ntrboot_flasher/master/README.md

windows (ex: windows 7 )
1) devkitARM r47
0-make folder(ex: c:\dev)
1-download
https://3dbrew.org/wiki/Setting_up_Development_Environment
2-devkitProUpdater-2.1.1.exe copy to (c:\dev)
3-localemulator install http://pooi.moe/Locale-Emulator/
4-devkitProUpdater-2.1.1.exe run with localemulator (japanese lang error)

2) firmtool
0-python 2.7.14 install
https://www.python.org/ftp/python/2.7.14/python-2.7.14.msi 
1-goto C:\Python27\Scripts
2-cmd (keyboard shift + mouse light "Open command window here")
pip install cryptography
pip install git+https://github.com/TuxSH/firmtool.git

3)firmtool.exe copy to 1)C:\devkitPro\msys\bin

4)git windows install
https://git-scm.com/download/win

5)make folder (ex:c:\work)

6)mouse light 
git bash here
git clone --recursive https://github.com/ntrteam/ntrboot_flasher.git

7)goto 1)C:\devkitPro\msys
run msys.bat

8)(ex:C:\work\ntrboot_flasher)
cd /c
cd work
cd ntrboot_flasher 
make

---------------------------------------------
Do not speak English well

Siul89, 2018-03-13:
i can't install firmtool, it says me this message
[image]

Siul89, 2018-03-13:
Ok, I compile it, this is the link 
https://mega.nz/#!hO4GBJLa!Xe-8W6-ysv_9EgACHjh4--nKbTrEyEvQzlEbKZjW1DM

closed 2018-10-31 by Siul89