ntrteam/ntrboot_flasher #85: Firmware flash check

wherearemylegs, 2018-01-21:
There are quite a bit of people in the Q&A that have had a misflash. They cannot tell that it hasn't flashed correctly until they try to perform ntrboot and it boots normally. Also, because the firmware was flashed, they cannot boot into their DS installer method, rendering their cartridge useless to them unless they know someone with a hacked 3DS. 

Can we implement a flash checker? Post-flash, it reads the flashed firm off and verifies that a proper flash occurred?

Bolt23, 2018-01-21:
@wherearemylegs I suppose ‘dsfirmverify.nds’ can do it.

wherearemylegs, 2018-01-21:
@Bolt23 Does that not require you to exit ntrflasher? Once exited, you cannot reenter any nds files using a misflashed cart. 

You could use dsfirmverify if you had a second cart but then you'd also be able to just re-flash the same cart

wherearemylegs, 2018-01-21:
Didn't realise ds_ntrboot_flasher and ntrboot_flasher were separate. Oops!

closed 2018-01-21 by wherearemylegs