ntrteam/ntrboot_flasher #86: Is the M3i Zero compatible with ntrboot_flasher

Gladuin, 2018-01-28:
I found out that I had that one laying around and saw that it was reverse engineerd [here](https://github.com/ntrteam/docs/blob/master/M3iZero-GMPZ003.md) and wondered if it would be compatible with this. It is the non GMP-Z003 version sadly

HandsomeMatt, 2018-01-28:
Only the GMP-Z003 would be compatible as it has no encryption. My repo has
the code and documentation that would make it possible to integrate into
flashcart_core.

Squirrel61, 2018-03-14:
If it's not too much work, I would greatly appreciate that, since I have a GMP-Z003 lying around here.

jason0597, 2018-03-25:
this is more of a [flashcart_core](https://github.com/ntrteam/flashcart_core) issue, not an ntrboot_flasher issue

handsomematt, 2018-07-09:
https://github.com/handsomematt/gmp-z003/tree/master/tools/z003_ntrboot

compile yourself - compatability is difficult, has to be GMP-Z003 and not on 4.5

closed 2018-07-09 by handsomematt