ntrteam/ntrboot_flasher #87: Compiling error.

thisismike97, 2018-02-16:
After following the compiling instructions i run into the following error:

$ make
make[1]: Entering directory `/c/Users/mike3/Desktop/GitHub/ntrboot_flasher/libncgc'
Makefile:26: *** missing separator.  Stop.
make[1]: Leaving directory `/c/Users/mike3/Desktop/GitHub/ntrboot_flasher/libncgc'
make: *** [/c/Users/mike3/Desktop/GitHub/ntrboot_flasher/libncgc/lib/libncgc.a] Error 2

Am i doing something wrong or is this a legit issue i've run into?

kitling, 2018-02-17:
hm. Try pulling submodules (`git submodule update --init`). I don't think that's the issue, but it's a possibility.

@angelsl since this is complaining about libncgc.

This is probably just a case where ntrboot_flasher needs to be updated and have a new release, though.

thisismike97, 2018-02-17:
Sadly pulling the submodules did not work. Still getting the same error as before.
(All commands were run from within the ntrboot_flasher folder.)

$ git submodule update --init

$ make
make[1]: Entering directory `/c/Users/mike3/Desktop/Github/ntrboot_flasher/libncgc'
Makefile:26: *** missing separator.  Stop.
make[1]: Leaving directory `/c/Users/mike3/Desktop/Github/ntrboot_flasher/libncgc'
make: *** [/c/Users/mike3/Desktop/Github/ntrboot_flasher/libncgc/lib/libncgc.a] Error 2

angelsl, 2018-02-17:
@thisismike97

Go to the libncgc folder and update it to the latest master, or remove the `undefine COMNFLAGS` line on line 26.

Alternatively, upgrade to a recent version of GNU Make.

thisismike97, 2018-02-17:
> Go to the libncgc folder and update it to the latest master, or remove the undefine COMNFLAGS line on line 26.

Both of the above worked, thank you for you fast reply.

> Alternatively, upgrade to a recent version of GNU Make.

Im using Make version 3.81 via msys included in the devkitpro. I dont really know how i should go about upgrading this since I installed the latest version of devkitpro yesterday.

In any case, my issue is now resolved.
Thanks.

closed 2018-02-17 by thisismike97

angelsl, 2018-02-18:
Make 3.81 now supported in 1e7e8efae07749878d76f90490290163b11ad5e6.