ntrteam/ntrboot_flasher #89: Acekard2i isn't restoring.

lunalik2, 2018-07-18:
Urgent: My Acekard2i isn't restoring. This same thing occurred on both my n3ds and o3ds. 
[image] Latest ver of ntr, and ntr flasher. I believe they are both on b9s 1.2, and sys ver 11.6.

kitlith, 2018-07-21:
could you upload your ntrboot.log?

maybe also increase the debug level a few times, try it again, and upload it?

However, I can't guarantee that I'll be able to help as it's been awhile since I've worked on this.

lunalik2, 2018-07-24:
[ntrboot.log](https://github.com/ntrteam/ntrboot_flasher/files/2225636/ntrboot.log)

kitlith, 2018-07-24:
hm, I guess the read failed?

iirc what's going on with the restore here is it's reading everything and making sure to only change what differs between the backup and the cart, because that's faster. you might be able to force it to write everything, but that involves modifying code. we also appeared to use a really large chunk size for this, reducing it to 0x10000 (don't go lower than this, though) could also help it progress more.

lunalik2, 2018-07-25:
Ok that would probably help. Thanks.