ntrteam/ntrboot_flasher #90: Issue with r4 sdhc card 2016 (Dual-Core / Real Time Save) from r4isdhc.com

SilkDeatH, 2018-08-08:
Hello. I have an issue with r4 sdhc card (Dual-Core RTS) from r4isdhc.com
It recognizes as r4isdhc family card in ntrboot_flasher, it creates backup files, it injects ntrboot. But when I reboot 3ds it wasn't recognizes as flashcart and does not react to START+SELECT+X+POWER combination while 3ds in the power off state... Please fix it.
[image]

kitlith, 2018-08-08:
One thing I notice is that you didn't mention any use of a magnet. Are you remembering to use a magnet to trigger the sleep mode switch while booting?

Unfortunately this project is essentially unmaintained at this point. In any case, for the sake of having things for the future... could you provide ntrboot.log from flashing it and before/after backups of the flash?

SilkDeatH, 2018-08-09:
Yup, I do it with the magnet using 3ds._.guide instructions.
In which mode I should create ntrboot.log (INFO/DEBUG/ERROR/WARN or NOTICE)?

SilkDeatH, 2018-08-09:
In it in the INFO mode, I always receive 1 string of text: [ERROR]: r4isdhc: found type 1 cart

kitlith, 2018-08-10:
So, it's possible to load the flashcart with Luma *when it's not flashed with ntrboot* unless it's an ak2i.

What about before/after backup.bin? That way we have a way to see what changed/if it injected properly.

SilkDeatH, 2018-08-10:
[ntrboot before dumping.log](https://github.com/ntrteam/ntrboot_flasher/files/2279427/ntrboot.before.dumping.log)
.log after dumping 27 mb
[ntrboot after injecting.log](https://github.com/ntrteam/ntrboot_flasher/files/2279434/ntrboot.after.injecting.log)
.log after restoring in progress, but I think it would be way too big..

kitlith, 2018-08-10:
I mean, uploading the backup from before the initial inject and a backup from after injecting.

SilkDeatH, 2018-08-11:
[r4isdhc-backup before, after injecting.zip](https://github.com/ntrteam/ntrboot_flasher/files/2280096/r4isdhc-backup.before.after.injecting.zip)