ntrteam/ntrboot_flasher #92: ReadMe File Must Inform the New Naming Convention for Existing Backups

emariz, 2018-08-19:
While the Release Notes for version 0.4 acknowledges that "Backups now have the shortname of the cart prefixed for backup/restore. e.g. ak2i-backup.bin", it does not state that existing backups must be renamed according to this new convention to be read by the flasher.
As it is currently phrased, that note only implies that backups are "created" with a short name, but says nothing about renaming existing backup files.

What is worse, the name possibilities that must be used to rename the existing backups are not stated in the ReadMe file or in the error message of the application (it only says that the restore file was not found.)
Because of the inexistent documentation, checking that the backup file is in the right location in the console's SD card or trying different versions of the corresponding backup are all in vain.

I checked the recent commits and noticed that the naming convention of backup creation (02cdb99, from Dec) and backup restoration (93f60c5, from Jan) had changed, but it is not easy to tell what exactly the new names must be just by reading those commits.

Luckily, I came across [this message](https://gbatemp.net/threads/i-have-a-problem-with-to-restore-my-r4i-flashcard.486642/#post-7882446) on GBATemp, and finally knew how to name my backup and re-flash my flashcard.

kitlith, 2018-08-19:
I updated the release text, but a list of possibl e names still needs to be put somewhere, maybe on the flashcart_core repo?

kitlith, 2018-08-22:
I'd much prefer the names be in the flashcart_core readme, listing which carts are supported by each name (and the short name to go along with it)

MRiCEQB, 2024-03-18:
Any updates on this?
I have a backup of a Ace3DS Plus as backup.bin that I want to restore with the latest flasher.