ntrteam/ntrboot_flasher #94: Add more confirmation to actions

SonoSooS, 2018-08-20:
I dumped the firmware of the flashcard, rebooted to GM9 to make a copy of the firmware, but accidently made a copy of the wrong file, so after flashing ntrboot to my flashcart I accidently pressed A twice and thus overwrote my only backup without any confirmation.

I'd suggest making a confirmation message for dumping the flashcart firmware if it was already dumped in the current session, and/or it already exists on the SDCard. This would save others the hassle of accidently permanently making their flashcarts unusable by accidently overwriting their only backup file. And let's face it, not many people think of making a backup (or dumping their firmware at all) before flashing ntrboot :)

...or just change the confirmation button from A to B when the progress reaches 100% so this could never possibly happen

emariz, 2018-08-21:
In its current state, does dumping firmware simply overwrite any existing dump? I thought that it added a number or a character, so that this would never happen.

And your flashcard is hardly ever "permanently unusable." Unless it is a unique specimen, most firmware revisions are already available on 3ds.guide.

kitlith, 2018-08-22:
@emariz it does not. That would require adding a menu to select which backup you wanted to restore.

@SonoSooS do I understand correctly that after performing the inject, backup was selected in the menu and you pressed A too many times, so it ended up preforming a backup?

SonoSooS, 2018-08-22:
@emariz yeah, it simply overwrites, without any confirmation, even if it exists or not. cancelling the backup will just zero out the file

also, luckily the flashcart I bought for my cousin happened to be a supported model, and lukily for me I checked the dump in G9 if it was valid, and the header bytes matched the one in the downloaded firmware, so this current flashcart I have is not lost yet I hope (I haven't yet tested because I have hardware difficulties doing ntrboot on the affected 3DS)

@kitlith after performing the inject, it asked me to press A to continue, but once I pressed A it started backing up the firmware. when I noticed it started backing up the firmware I turned the 2DS off in hopes that ntrboot_flasher would buffer the backup in RAM instead of writing it in chunks to the SDCard (like, why would a flashcart firmware be as big as 64Megs?), but sadly all it did is zero out the file. I blame the mushy keys of the 2DS.

kitlith, 2018-10-02:
This should be fixed in the next release (which should happen soon)

(`why would a flashcart firmware be as big as 64Megs?`: that's the size of flash, and we're dumping the whole thing 'just in case' even though most of it is (probably) unused.)

kitlith, 2018-11-27:
fixed as of 0.5.0

closed 2018-11-27 by kitlith