ntrteam/ntrboot_flasher #99: Looking for SPI flash image and/or FPGA bitstream for DSTwo

rfmerrill, 2019-07-14:
Hi! @hedgeberg told me to file an issue here for help with this.

I'm repairing a friend's DSTwo and wondering if anyone dumped the spi flash at any point while REing in case the flash chip is dead and I need to replace it. The FPGA bitstream would be nice too but that's probably a long shot (do they even field update it?)

kitlith, 2019-07-14:
I'm not aware of a dumped version of the bitstream. However, from what I remember of the update format, once decrypted it's basically a list of blocks to write to along with their data, that should be usable to restore (most of?) the flash.

I'll have to look/ask around to see if I can find the stuff I was working with when I was looking into this.

raisehero, 2021-01-28:
Hello, in my possession there's a working DSTwo with firmware 1.25. I'm not aware how to dump the flash without an external SPI flash clip tool. I might be willing into purchasing one of those, but not this month.