ntrteam/ntrboot_flasher #101: Console freezes after selecting my cart

Alftroz, 2020-12-14:
like the title says, i press a to skip the warning, and as soon as i select my cart, the console freezes and gets stuck
here's the log
[ntrboot.log](https://github.com/ntrteam/ntrboot_flasher/files/5686235/ntrboot.log)

hedgeberg, 2020-12-14:
Do you have a picture of the cart itself you could post? This is the kind of error we tend to run into when the cart is an unknown variant, or counterfeit, etc.

Alftroz, 2020-12-14:
https://media.discordapp.net/attachments/693992561064542278/787883880652734494/IMG_20201213_212833705.jpg
https://media.discordapp.net/attachments/693992561064542278/787883881453715466/IMG_20201213_212840214.jpg

hedgeberg, 2020-12-14:
Unfortunately, I don't think any of us have ever even _seen_ an r4 sdhc variant with a 2019 label on it. As such, this cart is not supported, and if I had to venture a guess, it won't be supported any time soon, as flashcart_core hasn't been worked on in a very long time. 

I don't want to tell you to buy a different cart, as I'm sure that would be frustrating, but for the forseeable future this cart will not work with ntrboot_flasher.

If you're feeling up to it, please see [flashcart_core](https://github.com/ntrteam/flashcart_core), read the rules for submitting a new flashcart, and provide the needed information so that we can gauge if it's worth adding support. If enough users end up buying this cart going forward, we may be able to justify returning to active development on this, but until then your best path forward will probably be to purchase a flashcart from the list of supported items. 

Marking this as closed for now.

closed 2020-12-14 by hedgeberg