ntrteam/ntrboot_flasher #103: Hope to add a forced write function

gba626, 2022-10-17:
I have a firmware corrupted r4isdhc.hk, ntrboot_flasher can't recognize it normally, it will prompt "Your flashcart is not supported.", when I solder the chip down and use the programmer to brush in the correct firmware, ntrboot_flasher can recognize it normally. If ntrboot_flasher can skip the type detection of flashcarts and force the firmware in directly, it can save a lot of trouble!

Squirrel61, 2022-10-24:
Hi, your post put me thinking. I still have 15 DualCore carts here that I bricked during the clone wars, by flashing a firmware that was supposed to be ok (according to the label) but it clearly wasn't. They were recognized by nothing, so there was no way to flash a working firmware to them. Also, the NDSL swap trick didn't work for these carts. So if I could create a version of the ntrboot_flasher with the security check removed, I could probably flash a working backup to these cartridges.

So I git cloned the repository to my Ubuntu virtual machine. And after many attempts, I managed to setup a working toolchain and compile a working ntrboot_flasher.firm. Once this succeeded, I tried removing the security check from the flasher. The result is the attached file. It does everything the original flasher does, including the "cartridge not supported" warnings. But instead of kicking you back to the flashcart selection menu, it just lets you pass through to the next menu.

I've tested this flasher and restored a known good working firmware backup to all bricked cartridges and was able to get 12 of them back to working order! I suspected already two carts of being hardbricked and it seems there was another hardbrick, which explains the 3 misses (I still have to check the hardware for bad solder joints).

So long story short, I've attached the modified flasher for you to try. For anyone reading this, BE VERY CAREFUL! You're guaranteed to brick your cart unless you know very well what you're doing! And don't bother the ntrteam with questions, because this modification was not created by them and they're not responsible for your bricks!
[ntrboot_flasher_mod.zip](https://github.com/ntrteam/ntrboot_flasher/files/9854930/ntrboot_flasher_mod.zip)

MushGuyDS, 2023-05-13:
Interesting mod that you made, @Squirrel61. Do you think there could be a possibility to force write on Acekard based flashcarts, whether it is to restore bricked flashcarts, or even to convert the R4i Ultra into an Acekard 2i (more info [here](https://www.reddit.com/r/3dshacks/comments/6ge48q/how_to_convert_an_r4i_ultra_r4ultracom_acekard/))? I tried restoring the R4i Ultra after a failed attempt at flashing with Decrypt9 resulted in a blank bootrom (it kept getting stuck at 12%), and trying to use ntrboot_flasher always resulted in getting stuck at 25%, so getting past that can be really helpful.

Squirrel61, 2023-05-13:
I think it's worth a try, since your R4i Ultra is currently bricked anyway. It's already some time ago since I made the mod and I can't recall exactly what I changed, but as far as I remember, the version check has been removed for all cartridges, not just the R4i Dual Core.

aarmastah, 2024-05-16:
> So long story short, I've attached the modified flasher for you to try. For anyone reading this, BE VERY CAREFUL! You're guaranteed to brick your cart unless you know very well what you're doing! [...]

Thanks for this! Using it (and a dump of my own card since he lost his), a friend was successfully able to un-brick his R4i Gold 3DS RTS Rev. D. Strangely, it's not listed as a "supported" card for ntrboot, so he wasn't able to re-flash it via traditional means.